orbbec/ros_astra_camera #190: Astra Pro Plus doesn't work when other cameras are connected

spok7, 2023-07-11:
I'm working with an Astra Pro Plus and several Intel RealSense cameras, and I kept getting the `Can't open /dev/bus/usb/` error. The issue was because uvc_camera_driver.cpp was trying to open one of the RealSense cameras, even when I specified the `serial_number`, the `uvc_vendor_id`, and the `uvc_product_id` in a roslaunch file.

This is because the two ID arguments were being ignored. UVCCameraDriver in src/uvc_camera_driver.cpp was trying to read them as ints, which failed since ROS1's yaml parser reads hexadecimal numbers as strings. This is why lines 104 and 105 of src/uvc_camera_driver.cpp need to be changed to:
```cpp
  config_.vendor_id = std::stoi(nh_private_.param<std::string>("uvc_vendor_id", "0x0"), 0, 16);
  config_.product_id = std::stoi(nh_private_.param<std::string>("uvc_product_id", "0x0"), 0, 16);
```

jian-dong, 2023-07-13:
Thank you for raising this issue. Fixed.

closed 2023-07-13 by jian-dong

BrendanWooAdmetal, 2024-06-19:
Newer versions of the code in uvc_camera_driver have re-created this issue by removing the effects of this commit.
It must be re-implemented in the newest code.