orbbec/ros_astra_camera #196: ros2 topic hz /camera/color/image_raw   is less than 4 Hz！！！！！

sun-rabbit, 2023-10-20:
Hello, I have been using Astro for RTAB-Map on a Jetson AGX Orin with Ubuntu 20.04 and an Astro Pro camera. For some reason, when launching examples, my image updates are very slow, and real-time synchronization is not possible. The maintainer of RTAB-Map suggested I check the Hz (frequency), and I found that my Hz is even lower than 4. Can you please tell me why this is happening? Is it a driver issue or an issue with the camera itself?
rtabmap[https://github.com/introlab/rtabmap_ros/issues/1054](url)

‘’‘
sun@root:~/ros2_ws$ ros2 topic hz /camera/color/image_raw 
average rate: 4.343
	min: 0.007s max: 0.495s std dev: 0.16495s window: 6
average rate: 4.901
	min: 0.004s max: 0.495s std dev: 0.14304s window: 12
average rate: 4.621
	min: 0.004s max: 0.495s std dev: 0.14736s window: 17
average rate: 4.768
	min: 0.004s max: 0.495s std dev: 0.13807s window: 23
average rate: 4.370
	min: 0.004s max: 0.559s std dev: 0.14701s window: 26
average rate: 4.580
	min: 0.004s max: 0.559s std dev: 0.14407s window: 32

sun@root:~/ros2_ws$ ros2 topic hz /camera/depth/image_raw 
average rate: 3.166
	min: 0.072s max: 0.741s std dev: 0.21469s window: 6
average rate: 4.134
	min: 0.017s max: 0.741s std dev: 0.20360s window: 12
average rate: 4.323
	min: 0.017s max: 0.741s std dev: 0.18391s window: 18
average rate: 4.546
	min: 0.017s max: 0.741s std dev: 0.17673s window: 25
average rate: 4.327
	min: 0.017s max: 0.741s std dev: 0.17370s window: 29

sun@root:~/ros2_ws$ ros2 topic hz /camera/color/camera_info
WARNING: topic [/camera/color/camera_info] does not appear to be published yet
average rate: 3.768
	min: 0.082s max: 0.367s std dev: 0.09341s window: 6
average rate: 3.784
	min: 0.030s max: 0.517s std dev: 0.14686s window: 10
average rate: 3.846
	min: 0.030s max: 0.517s std dev: 0.15045s window: 15
average rate: 4.185
	min: 0.000s max: 0.537s std dev: 0.16664s window: 21

sun@root:~/ros2_ws$ ros2 topic delay /camera/depth/image_raw 
average delay: 0.231
	min: 0.223s max: 0.237s std dev: 0.00602s window: 3
average delay: 0.237
	min: 0.216s max: 0.261s std dev: 0.01325s window: 8
average delay: 0.227
	min: 0.199s max: 0.261s std dev: 0.01641s window: 13
average delay: 0.228
	min: 0.199s max: 0.261s std dev: 0.01594s window: 17
average delay: 0.227
	min: 0.199s max: 0.277s std dev: 0.01988s window: 21
average delay: 0.229
	min: 0.199s max: 0.277s std dev: 0.02095s window: 25

sun@root:~/ros2_ws$ ros2 topic delay /camera/color/camera_info
WARNING: topic [/camera/color/camera_info] does not appear to be published yet
average delay: 0.049
	min: 0.039s max: 0.057s std dev: 0.00759s window: 3
average delay: 0.043
	min: 0.027s max: 0.062s std dev: 0.01260s window: 7
average delay: 0.046
	min: 0.027s max: 0.062s std dev: 0.01078s window: 11
average delay: 0.041
	min: 0.024s max: 0.062s std dev: 0.01120s window: 17
average delay: 0.040
	min: 0.024s max: 0.062s std dev: 0.01104s window: 21
average delay: 0.038
	min: 0.024s max: 0.062s std dev: 0.01050s window: 27

’‘’

sun-rabbit, 2023-10-20:
@tfoote @lhongzhong

6APA6ALLlKA, 2023-11-26:
@sun-rabbit  hello, please, can you tell me how to calibrate astra pro plus on ros2? i cant view color images, only ir and raw. what packages do you use and which ros2 library i should install?